Z80-Retro/2063-Z80 #3: Allow alternate SD card sockets

johnwinans, 2021-11-08:
Due to the galactic parts-shortage, Digikey, Mouser, and Newark are all on backorder for the Hirose DM3D-SF SD card socket in the Rev 3 design.

If a Rev4 board is produced, adjust the footprint to allow for an alternate socket (such as the Molex 1051620001) to be used.
[It looks like the Molex could be made to fit on the same pads IFF the keep-out areas are increased and a second card-det landing is added.]

sebsch2000, 2021-11-17:
Amphenol 114-00841-68 has the same footprint as the Hirose part. Right now Mouser has 4.813 of them in stock (Mouser part no. 523-114-00841-68).

jackrubin, 2021-11-18:
Who’s on first? Does this mean that there is no need for a Rev 4 board spin and I can build out one of your Rev 3 sets?

Thanks,
Jack

sebsch2000, 2021-11-18:
Well, I built a rev 3 board this week. I ordered the parts from Mouser and Digikey. Had to replace the Hirose connector with the Amphenol one and the ICL3232CPZ (U12) with the MAX232N which is basically the same. The test programs worked as they should.

jackrubin, 2021-11-18:
That’s good – if you still have a spare, I’d like to buy one board + programmer set from you. Please let me know the cost and PayPal info.

Thanks,
Jack

johnwinans, 2021-11-18:
How did I not find the Amphenol 114-00841-68 part????  That is fantastic!
I'll add a note marking that as an alternate and take a pass on the footprint hacking.

jackrubin, 2021-11-26:
Please push the rev/bom changes up to the github repository when you get a chance.
Thanks!

johnwinans, 2021-12-01:
There is at least one alternate that does not require a footprint modification.  I am not going to leave well enough alone.

closed 2021-12-01 by johnwinans