Z80-Retro/2063-Z80 #9: Is board silkscreen for U11 incorrect on 2063.4?

joelburton, 2022-03-03:
The board silkscreen on U11 is `'00` (suggesting a 74HCT00) --- however, the schematic/values/BOM suggests that is a 74HCT132.

Is this just a mislabel on the board? Or am I misunderstanding something significant?

johnwinans, 2022-03-03:
The 132 is the correct part!  The reason is that U11D must have a 
Schmitt trigger.

Thanks for mentioning it. I will update the silkscreen.

I recorded a video explaining the need for the Schmitt here:

https://youtu.be/LDce9foQ2Ag

I have been working on a playlist discussing every detail of this 
project here:

https://www.youtube.com/playlist?list=PL3by7evD3F51Cf9QnsAEdgSQ4cz7HQZX5

This month I will be adding details on how to port CP/M onto it.

-John

johnwinans, 2022-03-22:
I updated the silkscreen.  It will appear in v4rc2

closed 2022-03-22 by johnwinans