Z80-Retro/2063-Z80 #10: Possibly useful advice: connecting console to computer w/o RS232 serial

joelburton, 2022-03-21:
I have a RS232 terminal, but a friend building the board does not and doesn't even have a serial port on her computer. The design doesn't expose an easy way to connect a 5v Tx/Rx for a FTDI->USB connection.

I doubt there's space to add that to the board (maybe above the printer connection?), but something that is working for us is a RS232 voltage -> 5v converter (we're using https://www.amazon.com/gp/product/B07LBDZ9WG/). This can be connected to a standard FTDI USB cable.

johnwinans, 2022-03-21:
This is a good idea... and it figures... I *just* ordered some v4 boards last week to verify that design. :-D

I don't have a serial port on mine either.  I use a USB/RS232 adapter like one of these:
https://www.amazon.com/Sabrent-RS-232-Adapter-Prolific-SBT-USC6K/dp/B000A1KD1K/ref=sr_1_10?crid=EX09T7WCM2CL&keywords=rs232+usb+adapter&qid=1647901137&s=electronics&sprefix=rs232+usb+adapter%2Celectronics%2C66&sr=1-10

But this makes me think.  There are likely a number of potential users the might want to go USB-direct.

With the product you mention, you could solder in wires to the U12 pin holes and run them to the adapter board. With the header on the side next to the DB25 seems awkward to try to directly mate.  (That is actually how I did the first version of this board as I messed up U12 completely.)

If you leave U12 unpopulated, we could add a 1-row header that could match an available 3rd party USB TTL adapter.  These appear to be pinned the same way.  
https://www.sparkfun.com/products/14050
https://www.sparkfun.com/products/9716

I'm not sure that I'd want to power the Retro board from a PC USB port... and would definitely not think of powering a PI (FLASH programmer) from it.  (I *always* Power my retro FROM the PI when I have the FLASH programmer installed.)  But it might be worth providing a way to add a jumper or something to allow it.

It should be easy to put a header just south of the printer port or even under U12 (so you could either install a header OR the U12 socket.)
If you install a right-angle header then the adapter would lay flat (which would be my preference.)

What do you think?

joelburton, 2022-03-21:
Putting the headers in the center of U12 is an interesting idea an could help those that don't populate it; that wouldn't use up any of the precious board space :)

joelburton, 2022-03-21:
Those Sparkfun boards are what a lot of other projects (like the rc2014s) make their pins compatible for, so a lot of people have them lying about.

johnwinans, 2022-03-22:
I was torn between the header at the bottom under the printer port and overlaying U12.
I like the idea of building the board with both the U12 socket and J13 installed so that one can flip back & forth between using TTL and RS232.

What do you think of this PCB arrangement?

https://github.com/johnwinans/2063-Z80/blob/add-ttl-serial-header/2063-Z80.jpg

joelburton, 2022-03-22:
Looks awesome to me!

johnwinans, 2022-03-22:
Coolness!  I will make it happen and merge it in tonight!

closed 2022-03-23 by johnwinans