Z80-Retro/2063-Z80 #16: Invalid FLASH signature: 0x2020 != 0xbfb5

Steve-Sexton, 2022-09-17:
I've flashed several files to the retro board with no problems.  

Last time I tried to flash something to the board I get:
"Product ID: 0x20, 0x20"
"Invalid FLASH signature: 0x2020 != 0xbfb5"

Any suggestions would be grateful.

johnwinans, 2022-09-17:
What happens if it is not plugged into the z80 board?

What rev of the programmer board do you have? Does it have pullups on the 23017 reset inputs?

Steve-Sexton, 2022-09-17:
I'm using  Programmer Rev 2065.2 and  Z80-Retro Rev 2063.4.2

Working now, something must have been in an unknown state.

Unplugged the ribbon cable from the Programmer.
Attempted to flash blinky1
RPI reported many flach.c mcp23017_write errors
Unplugged the Programmer J3 - PI power jumper
Plugged the ribbon cable back into the RPI
Plugged the jumper across Programmer J3 - PI power
Flashed blinky with no errors.... LED in blinking

Thanks

johnwinans, 2022-12-08:
I'm glad to know this case is closed!

closed 2022-12-08 by johnwinans