Z80-Retro/2063-Z80 #17: Is this a Raspberry pi extension project. or z80 cp/m

Amstrad-nutter, 2022-11-09:
A Raspberry PI project?
As you cannot seem to Communicate with z80 Repo, without one.
Compile the Files, without one
Program the flash, without one
Create as SD card image, without one
Please for us that use the Precious Pi’s we have running Infrastructure e.g., Home Assistant an MQTT Server etc.
Can you just run MAKE and zip the resulting files. and end the frustration.
So many of us who have Device programmers and dormant z80 retro’s running nop loop tests can join in the fun too.
Please Help us.
Many thanks.

johnwinans, 2022-12-08:
I am not sure what you are proposing.
You can run the build and SD-card programming on any Linux box.
You can also program the FLASH with any suitable programmer.
No PI is needed.

johnwinans, 2023-02-11:
I just re-read this.... apparently I completely missed the request for binaries in there.
OK, I'll build a release with binaries for this project and the related CP/M part of the project.

johnwinans, 2023-02-11:
See version V4.3.3 with fresh PDF and binaries in the attached zip file

closed 2023-02-11 by johnwinans