Z80-Retro/2063-Z80 #20: Documentation of Change of C1, C11, C12 from 0.1uF to 0.47uF on 2063-Z80-4.4.0

dennowiggle, 2023-05-06:
C1, C11, C12 (not C2) around U12 ICL3232CPZ on the 2063-Z80-4.4.0 board changed from 0.1uF to 0.47uF on the schematic and BOM but this is not indicated in the notes on the github page for this version of board, or in the schematic delta/diff file, or in the accompanying video about the changes. It would be nice to update documentation.

I would guess this change is to match the datasheet values for 5.0V operation.

Thanks

johnwinans, 2023-05-06:
Well, the diff images that I recently added are between tags: 4.3.3 and 4.4.0  so the are correct.

But it might have been a good idea to also generate a 4.3.2 to 4.3.3 diff as well... or maybe a 4.3.2 to 4.4.0 diff.

There are also some v3 to v4 diffs in there... that are probably pointless since v4 does not actually exist as a tag.

johnwinans, 2023-05-06:
FYI - I just added V4.3.2-V4.3.3-diffs.sh and the related diff images.  That should make things a bit more clear as to what is going on since you are referring to changes over two different releases rather than changes across only one release.  I would argue that the real problem is that I failed to generate a proper diff when I released v4.3.3.  Sorry.

I also cleaned up the old poorly named script: v3-v4-diffs.sh and renamed it to what it actually does which is to generate a diff between the v3 tag and the HEAD branch... which is not vary useful other than for local testing.

closed 2023-05-06 by johnwinans

dennowiggle, 2023-05-06:
Thank you for the clarification. I know how much work all of this is, for a hobby.